earlephilhower/arduino-pico-littlefs-plugin #6: arduino-pico-littlefs-plugin  Release 0.2.0 not working on ide 2.0 latest

karadevnet, 2023-03-05:
arduino-pico-littlefs-plugin  Release 0.2.0 not working on arduino ide 2.0.4 latest download
i try to add in arduino ide 2.0.4 to can use a little fs but upload in tools menu not show up or have some other way to make it.

also have a question about if it is possible to make 128kb - 1Mgb separate from space of memory visible from windows like a flash drive to can upload/download/edit files in storage and change a file inside . if it is possible i can make my own file explorer to work with this tiny little fs flash drive for save/edit different files for different applications of pico board. will be cool
best regards.

earlephilhower, 2023-03-05:
See #3.  IDE 2.0 doesn't support plug ins.

closed 2023-03-05 by earlephilhower

earlephilhower, 2023-09-29:
Please see the [arduino-littlefs-plugin](https://github.com/earlephilhower/arduino-littlefs-upload) for use under IDE 2.2.1 or later