earlephilhower/arduino-pico-littlefs-plugin #8: arduino-pico-littlefs-plugin in Arduino 2.1.1

Gavin-Perry, 2023-08-01:
Does arduino-pico-littlefs-plugin work with 2.1.1? Seems like it has been long enough!
If not how do I use LittleFS (got 1m & 1m) and get data onto the  Pico?
I tried downloading but not sure what install means since what I get from unpacking the zip file is nothing like what is shown in the example.
Following the folders all the way down I have:
C:\Users\gavin\OneDrive\Documents\Arduino\tools\arduino-pico-littlefs-plugin\bin\com\efp3\pico\mklittlefs
I thought about the make.sh file, but I'm on Windoze, not linux or unix.
Thanks for all the help and code you do.

earlephilhower, 2023-08-01:
No, the new Arduino IDE has no support for plug ins.  See https://github.com/arduino/arduino-ide/issues/58

closed 2023-08-01 by earlephilhower

Gavin-Perry, 2023-08-02:
So how can I use the LittleFS and put files onto the PICO or PICO W?

Sent via the Samsung Galaxy S22 5G, an AT&T 5G smartphone

earlephilhower, 2023-08-02:
Either use the 1.x IDE or run the `mklittlefs` and `u2fconv.py` scripts manually.  You might just use the 1.x IDE only for littlefs uploads but keep the 2.x for your other work, for example.

Gavin-Perry, 2023-08-02:
It's surprising that they would not support tools in version 2. This workaround will have to do.
Thanks for your help.

Gavin

Sent via the Samsung Galaxy S22 5G, an AT&T 5G smartphone

per1234, 2023-08-03:
>It's surprising that they would not support tools in version 2

Extending the capabilities of Arduino IDE is supported. However, the system for doing so is very different from [the "Tool" system Arduino IDE 1.x inherited from the Processing IDE](https://github.com/processing/processing4/wiki/Tool-Overview) used by this project.

Arduino IDE 2.x is built on the [**Eclipse Theia** IDE framework](https://theia-ide.org/). Since it is based on VS Code, [Theia supports VS Code extensions](https://theia-ide.org/docs/authoring_vscode_extensions/), and so Arduino IDE 2.x has supported these extensions from the start (and they are already heavily used by Arduino IDE 2.x). VS Code extensions is the form the next generation of Arduino IDE "plugins" will take.

In addition to the framework's support for extensions, such "plugins" also need access to data from the Arduino framework. The lack of access to such data blocked the possibility of creating VS Code extension replacements for the Arduino IDE 1.x "plugin" tools. That missing capability was recently added:

https://github.com/arduino/arduino-ide/pull/2110

So there is now no technical barrier to creating an IDE 2.x replacement for this project. You can see a replacement for the popular "ESP Exception Decoder" Tool was already created:

https://github.com/dankeboy36/esp-exception-decoder

Gavin-Perry, 2023-08-03:
Cool!
So there is hope for a way to easily put files into the file system on a Pico when the program is loaded.
Surely there must be a way to put files there at some point or there would be no point in having a file system.
What am I missing?

Sent via the Samsung Galaxy S22 5G, an AT&T 5G smartphone

arduino/arduino-ide#2110<https://github.com/arduino/arduino-ide/pull/2110>

https://github.com/dankeboy36/esp-exception-decoder

earlephilhower, 2023-09-29:
Please see the [arduino-littlefs-plugin](https://github.com/earlephilhower/arduino-littlefs-upload) for use under IDE 2.2.1 or later

Gavin-Perry, 2023-09-29:
Thanks for the littleFS. That's great work!

Gavin