earlephilhower/arduino-pico-littlefs-plugin #12: Question regarding littleFS

hasenradball, 2024-07-26:
@earlephilhower 
when using / needed a filesysten would you usecthe arduino framework for the pi pico or the c sdk of the Pico?

Best regards 
Frank

earlephilhower, 2024-07-26:
Unless something has changed that I'm not aware of, the SDK has no filesystem support.  You can definitely roll your own LittleFS implementation, they make it relatively easy to do.  You'd be calling the `lfs_XXX` calls natively, not `fopen()` or whatever, but there's almost a 1:1 mapping.

This plugin is only useful for the Arduino Core, of course, since it depends on the memory and linker files present on the Pico (and ESP32 and ESP8266).  `mklittlefs`, the app which makes the filesystem bitstream for upload, can be used for roll-your-own LittleFS as long as you set the parameters appropriately and run it manually.  Uploading to the Pico after doing so it a problem left for the student. :laughing:

closed 2024-07-26 by earlephilhower

hasenradball, 2024-07-26:
Thanks for your answer.
I am comming from the Esp family with the arduino core.
As I started with the pico, first started with micropython but from the bottom of my heart I like c/ cpp.

But it seems easier to stick on the arduino core than using the c sdk for the pico.